drtyhlpr/rpi23-gen-image #146: Ethernet not working or RPi 3B+

BrainStone, 2018-11-06:
The image generates just fine and works out of the box, but the ethernet is not working.

While digging around to find a working 64-bit version for my Pi, I came across several pages stating, that the ethernet would not be working due to some outdated drivers, but it should be resolved in a future release.  
Would you be able to look into this to make sure the right drivers are downloaded for the 3B+?

I sadly don't remember where I have seen this. But it was multiple different pages for multiple different distributions.

burnbabyburn, 2018-11-06:
I am building this for 1-2 weeks now and never got a problem with wired ethernet. Can u post your buildconfig/template?

BrainStone, 2018-11-06:
That's my template

```sh
RPI_MODEL=3P
RELEASE=stretch
BUILD_KERNEL=true
KERNEL_ARCH=arm64
RELEASE_ARCH=arm64
CROSS_COMPILE=aarch64-linux-gnu-
QEMU_BINARY=/usr/bin/qemu-aarch64-static
KERNEL_DEFCONFIG=bcmrpi3_defconfig
KERNEL_BIN_IMAGE=Image
KERNEL_IMAGE=kernel8.img
KERNEL_BRANCH=rpi-4.11.y
ENABLE_WIRELESS=true

# Custom settings
APT_INCLUDES=console-setup,keyboard-configuration,net-tools,systemd-sysv,wpasupplicant
APT_SERVER=ftp.de.debian.org
DEFLOCAL=de_DE.UTF-8
ENABLE_CONSOLE=false
ENABLE_REDUCE=true
HOSTNAME=TESTING
REDUCE_HWDB=false
REDUCE_LOCALE=false
REDUCE_SSHD=false
```

And I use DHCP.

burnbabyburn, 2018-11-06:
Main differences to my config:
`RELEASE=stretch`
and
`ENABLE_IFNAMES=false` to get the old naming back.

Maybe thats the problem? does /usr/lib/systemd/network/10-eth.network point to eth0 and your interfaces has the same naming?

BrainStone, 2018-11-06:
I also have `RELEASE=stretch` or did you mean that you don't use stretch?

And `ENABLE_IFNAMES=false` would mean it'll be called `eth0`?

The other stuff I can't check right now. I don't have access to the Pi.

burnbabyburn, 2018-11-06:
i use buster. But i don't think thats the issue here. I think the systemd service points to eth0. This is fine if old naming is used. 

To actually enable the old naming ENABLE_IFNAMES=false is correct:
`Enable automatic assignment of predictable, stable network interface names for all local Ethernet, WLAN interfaces. This might create complex and long interface names. This parameter is only supported if the Debian releases stretch or buster are used.`

TBH i never tried the new naming scheme so it's possible its broken.

BrainStone, 2018-11-06:
Ok. I'll bump to buster and add `ENABLE_IFNAMES=false`. Mostly because I want `eth0` back xD.

I'll let you know tomorrow when I'm working on it again.

Is there a decent place to chat about this project? Like a Discord or IRC? I have some ideas I might turn into PRs if they are wanted.

burnbabyburn, 2018-11-06:
OK!
But please check if thats actually the problem before flashing a new image ;) 
If my guess is correct, we should improve that part

BrainStone, 2018-11-07:
> Maybe thats the problem? does /usr/lib/systemd/network/10-eth.network point to eth0 and your interfaces has the same naming? Didn't see any code to change systemd from default eth0 

`/usr/lib/systemd/network/10-eth.network` doesn't exist but `/lib/systemd/network/10-eth.network` does.  
The content of the file is

```toml
[Match]
Name=

[Network]
DHCP=yes
```

I sadly can't check `ifconfig` because that wasn't installed on that image. (Should be one of the default packages really)

BrainStone, 2018-11-07:
Ok. Even with the new image it's not working.  
I also can't get WiFi to work. The wpa_supplicant service can't seem to connect to it.

I could bet it's a tiny thing that's missing here.

BrainStone, 2018-11-07:
I did some research and found these threads on the topic. Though they all stay a bit vague...

- https://www.raspberrypi.org/forums/viewtopic.php?t=211018
- https://mail-index.netbsd.org/current-users/2018/08/25/msg034111.html

It looks like `CONFIG_USB_LAN78XX=y` is missing in the kernel config. Even in the `/boot/config-rpi-4.11.y` file it says that this config has no value set (Exact line: `# CONFIG_USB_LAN78XX is not set`).

What do I need to do, to configure that?

Btw you said you were successfully using this project for a few weeks. Though keep in mind that the 3 B+ has an Ethernet chip that's different to all other previous chips.

burnbabyburn, 2018-11-07:
I always use 4.18. I am scared of current branch. 
i compiled your config, booted up on a RPI3B+ and got eth0 dhcp auto configured :)

BrainStone, 2018-11-07:
With `rpi-4.11.y`? Or `rpi-4.18.y`?

burnbabyburn, 2018-11-07:
> 
> 
> With `rpi-4.11.y`? Or `rpi-4.18.y`?

4.18

`The wpa_supplicant service can't seem to connect to it.`
This is a configuration problem:
with wireless=enabled systemd is taking over control of wlan0. You need to get systemd using wpa_supplicant

```
#WLAN-CLIENT-MODE

echo "
ctrl_interface=/var/run/wpa_supplicant
eapol_version=1
ap_scan=1
update_config=1 " >> /home/pi/pihole
--------------------------------------------------------------
# wlan client

echo "network={
        ssid="mySSID"
        scan_ssid=1
        pairwise=CCMP
        group=TKIP
        proto=RSN
        key_mgmt=WPA-PSK
        psk="mypsk"
}
" >> /etc/wpa_supplicant/wpa_supplicant-wlan0.conf

systemctl enable wpa_supplicant.service
systemctl enable wpa_supplicant@wlan0.service

systemctl start wpa_supplicant.service
systemctl start wpa_supplicant@wlan0.service
```

you will need `crda `package for wifi (wlan freq and power output db for not breaking local law)

BrainStone, 2018-11-07:
The 4.14 kernel is enough already.

I'll make a PR tomorrow bumping the default kernel version in the templates.

closed 2018-11-07 by BrainStone