drtyhlpr/rpi23-gen-image #148: Why is systemd-sysv actively removed?

BrainStone, 2018-11-07:
I noticed that the package `systemd-sysv` is actively removed when using buster or stretch: https://github.com/drtyhlpr/rpi23-gen-image/blob/95ed69af46701df47875794c0dd4d6339c0c51a8/bootstrap.d/10-bootstrap.sh#L24

I noticed the comment above that says
> Exclude packages if required by Debian release
though I'm not sure what exactly that means. Does it mean that Debian mandates these packages to be removed?

burnbabyburn, 2018-11-07:
Don't know the reason, but i like just having init OR systemd. Mixing up both is like Raspbian. 

P.S. i can't find systemd-sysv on my "buildbox" installed wih debian testing. Maybe that exclusion is not even neccessary (anymore)

BrainStone, 2018-11-07:
I mean without it the commands like `reboot`, `poweroff`, `shutdown` are all missing and I have to revert to `systemctl reboot`.

From what I understand `systemd-sysv` is the package that provides these.

burnbabyburn, 2018-11-07:
I am building your config atm. If you remove that exclude you get systemd-sysv automatically back.
Your totally right with the commands above

```
/sbin/halt
/sbin/init
/sbin/poweroff
/sbin/reboot
/sbin/runlevel
/sbin/shutdown
/sbin/telinit
...
```

BrainStone, 2018-11-07:
I mean it's not urgent to have that package, though I'd love to have it eventually, and I think it should be installed by default.

So maybe add a config for it? Or remove it altogether. Don't really see a purpose in removing these packages.

And is it really included by default without having to specify it in `APT_INCLUDES`? Because in my config it is there.

burnbabyburn, 2018-11-07:
+ . bootstrap.d/10-bootstrap.sh
+ . ./functions.sh
+ VARIANT=
+ COMPONENTS=main
+ EXCLUDES=
+ [ false = true ]
```
+ [ false = true ]
+ http_proxy=http://127.0.0.1:3142/ debootstrap --arch=arm64 --foreign --components=main --include=console-setup,keyboard-configuration,net-tools,wpasupplicant,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils,......
```
If you remove the above mentioned exclude its included per default

burnbabyburn, 2018-11-07:
i should go back to bed today...
`+ http_proxy=http://127.0.0.1:3142/ debootstrap --arch=arm64 --foreign --components=main --include=console-setup,keyboard-configuration,net-tools,wpasupplicant,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils,locales,keyboard-configuration,console-setup,parted,dbus,openssh-server,alsa-utils,rng-tools,libnss-systemd stretch /home/helge/rpi23-gen-image/images/stretch/build/chroot http://ftp.de.debian.org/debian
`

burnbabyburn, 2018-11-07:
Jap, i added ENABLE_SYSVINIT for this. Set it to ENABLE_SYSVINIT=true to get the desired packages. Default behaviour is still to remove them. Dirty Helper has to decide his defaults

BrainStone, 2018-11-07:
Ah cool. I'm hoping this will get merged soon. Don't want to change the base repo of my submodule...

burnbabyburn, 2018-11-09:
It can be added to buster. I didn't test jessie or stretch though

burnbabyburn, 2018-11-16:
#150 is merged this issue can be closed

drtyhlpr, 2018-11-16:
perfect

closed 2018-11-16 by drtyhlpr