drtyhlpr/rpi23-gen-image #149: Custom default configs

BrainStone, 2018-11-07:
Is there an easy way to provide some customized default configs?

Like for example I'd like to provide a `/etc/wpa_supplicant/wpa_supplicant.conf` file so that the Pi can connect to the WiFi straight away.

burnbabyburn, 2018-11-07:
1. Replace  
/files/network/wlan.network
with your config file

or better

add the possibility to configure it into the script ;)

BrainStone, 2018-11-07:
wpa_supplicant is a totally different file format.

The `custom.d` folder is supposed to be in the repo root, right?  
Because I think I'll just throw a script in there that'll write that file to the right place.  
And if you don't mind answering, what is the variable for the `/etc` or `/` dir of the image?

burnbabyburn, 2018-11-07:
I didn't get this right the first time :)
You want to configure Wifi via wpa_supplicant not systemd.

Probably the best way to include your personal files is a custom script.

> And if you don't mind answering, what is the variable for the /etc or / dir of the image?
${ETC_DIR} = /etc dir in build_chroot
${R} = root dir in build_chroot
${LIB_DIR} = /lib dir in chroot

This is what i understand from reading the code

burnbabyburn, 2018-11-07:
BTW for custom scripts there is the following snipet

#Scripts in the custom.d directory will be executed after all other installation is complete but before the image is created.
## Execute custom bootstrap scripts
if [ -d "custom.d" ] ; then
  for SCRIPT in custom.d/*.sh; do
    . "$SCRIPT"
  done
fi

BrainStone, 2018-11-07:
That's what I'm using and it's working great.

closed 2018-11-07 by BrainStone