drtyhlpr/rpi23-gen-image #152: KERNEL_CCACHE breakes kernel compilation

BrainStone, 2018-11-08:
Hi. I'm not sure if someone has had that issue before but it seems like that enabling `KERNEL_CCACHE` creates broken kernels if I am copiling different versions for testing.

I noticed this because suddenly my Pi wasn't getting an IPv4 connection anymore over ethernet. I tested it back with an older build I had handy and compiled before trying out different versions, which worked flawlessly. Then I recompiled the exact same build after I had compiled two different versions of the kernel and it stopped working, like with the other versions.

Has someone made similar experiences?

BrainStone, 2018-11-08:
Ok. I'm at a complete loss now. Even turning `ccache` off, still produces a broken image...

burnbabyburn, 2018-11-08:
did you install ccache package on host machine?

BrainStone, 2018-11-08:
Ok. I'm at a complete loss now...

It must've been something else. I also cannot find anything significant that's different. No errors etc.
Only real difference I could find was that the build that worked cloned the repo, while the other just fetched it.

If that doesn't work, would uploading the build logs help?

burnbabyburn, 2018-11-08:
If i understand right. You did change the setup to local repos? (aka clone firmare,linux-kernel etc.) after that it's not working with ccache or without ccache?

If that's correcet check the following: you checked out master branch of linux kernel (which is 4.14 atm) and told your template u would like to use 4.18?

BrainStone, 2018-11-08:
I expressed myself badly.

I have a git repository which I use to build my images. It contains all the settings and templates and additional bits and bobs. It also includes this repo as a submodule.

Now I use CI to build the actual images. And the only real difference I can spot between the working image and the not working image is the fact that the working image was build on a repo that was cloned and the not working was built on a repo that was merely updated.  
But since I've managed to get a few other builds with a cloned instead of a updated/fetched repo I kinda doubt that's the issue.

Another thing is that the backing mirror I'm using changed, as the original behind `ftp.de.debian.org` is still offline, so they're using a different one. Though I'd be surprised just as much if that made the difference.

BrainStone, 2018-11-08:
As I supected it makes no difference between cloning and updating/fetching...

BrainStone, 2018-11-08:
And any kernel version is not working in the same way that the interface is not receiving an IPv4 address and using a self assigned IPv6 (because it's not communicating with the DHCP).

If it helps I can upload the working and not working image plus their respective logs.

burnbabyburn, 2018-11-08:
Well... i have not the slightest clue what your error/problem could be :)
Pls post build.log maybe there is a hint.

Second: Do you use TravisCI? Is it possible to see the build script? I tried it yesterday but TravisCI Ubuntu LTS 14.04 is not able to find crosscompiler package

BrainStone, 2018-11-08:
I use the built in CI of GitLab on a shell runner which runs the script as a user on a local machine.

Job1 is the one that works and Job2 is the one that doesn't.

[job1.log](https://github.com/drtyhlpr/rpi23-gen-image/files/2562455/job1.log)
[job2.log](https://github.com/drtyhlpr/rpi23-gen-image/files/2562456/job2.log)

burnbabyburn, 2018-11-08:
I don't see any major difference.
job1.log clones the kernel
job2.log uses tarball?
```
Removing 2018-11-08-arm64-rpi-4.18.y-rpi3P-buster-arm64.img.xz
HEAD is now at 58d3950 Update Kernel to 4.18
```
I don't think it's an image problem ;)

I'll qoute myself:

> Did you check the systemd file

BrainStone, 2018-11-08:
Where does job2.log use a tarball?

And the log line `Removing 2018-11-08-arm64-rpi-4.18.y-rpi3P-buster-arm64.img.xz` is just cleanup from a previous build (One for kernel 4.18 naturally).

I checked that the name is in the file. `ifconfig` even lists `eth0`. But just with an IPv6 adress.

burnbabyburn, 2018-11-08:
```
root@pihole:~# cat /usr/lib/systemd/network/10-eth.network
[Match]
Name=eth0

[Network]
DHCP=yes
```
This file looks the same?

burnbabyburn, 2018-11-08:
I've finally build the Image on RPI3B+ 
still got eth0 with DHCP up and running after boot. But it seems i need to fix DNS everytime.

drtyhlpr, 2018-11-09:
hi brainstone. thanks for you feedback and thanks again burn for your amazing support!
ccache was recently added and I think I mostly used it to compile and run qemu kernels without realy using the virtual Ethernet functionality.

network problems: if possible use tcpdump -vvv or tshark/wireshark to get a more detailed idea of the problem. is the phy link up? is the interface configured with dhcp? (what do you see in the logs of your dhcpd), have you tried to configure the interface with static ip? do you use ipv4 or ipv6? can you ping your own ip, can you ping the gw ip? does dns name resolving work?

I have to admit that I don't know what CI is and need to findout. But the point is: (if you havent so far) please use the exact configuration template that breakes your ethernet and build it on a regular linux command shell (linux vmware) and compare if the the ethernet errors are still is the same.

ps. about "I noticed this because suddenly my Pi wasn't getting an IPv4 connection anymore over ethernet." does this mean with the image you created you get and IPv4 connection but then loose it aggain? or does this mean suddenly form one day to another after you build the image IPv4 wasnt possible anymore?

BrainStone, 2018-11-10:
> if possible use tcpdump -vvv or tshark/wireshark to get a more detailed idea of the problem.

Currently not possible. I don't have access to the machine for a while now sadly. Might mean this issue will become stale for a little while. Though I'll come back for it for sure.

> is the phy link up?

Yes. `eth0` is up and has an IPv6 adress.

> is the interface configured with dhcp?

Yes

> (what do you see in the logs of your dhcpd)

Can't check for now. Sorry :(

> have you tried to configure the interface with static ip?

No. 

> do you use ipv4 or ipv6?

I typically use IPv4

> an you ping your own ip, can you ping the gw ip?

I ping the IPv6 of the Pi from itself and any other device on the network. I can't ping any IPv4s from the Pi.

> does dns name resolving work?

No.

> I don't know what CI is

CI stands for "Continuous Integration". Which is a fancy term for automated building. In other words whenever I make changes to my repo, a build script is executed. And in this case it happens to make an image for me.

> please use the exact configuration template that breakes your ethernet and build it on a regular linux command shell (linux vmware) and compare if the the ethernet errors are still is the same.

I will try that, though I doubt it'll make any difference as the CI is also just a shell in essence.

> does this mean with the image you created you get and IPv4 connection but then loose it aggain? or does this mean suddenly form one day to another after you build the image IPv4 wasnt possible anymore?

It means that from one day to the other the image stopped communicating with DHCP and receiving an IPv4 automatically. At no point is a working IPv4 connection established from what I can tell.

burnbabyburn, 2018-11-14:
p.s. maybe have a look here:
https://github.com/Crazyhead90/pi64

BrainStone, 2018-11-15:
I just confirmed that it's not the fault of the CI runner. I built it on an independent and still the same.

BrainStone, 2018-11-20:
Ok. More debugging:

- A static IPv4 works flawlesly.
- When booting there are no logs for the DHCP client, weirdly enough (With no static IP and DHCP enabled!)
- Manually starting the client with `dhclient` yields this log:  
  ```
  Nov 20 17:43:56 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
  Nov 20 17:43:56 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 8
  Nov 20 17:43:59 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
  Nov 20 17:44:03 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 4
  Nov 20 17:44:04 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 11
  Nov 20 17:44:07 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 10
  Nov 20 17:44:15 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
  Nov 20 17:44:18 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 10
  Nov 20 17:44:22 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 7
  Nov 20 17:44:27 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 9
  Nov 20 17:44:29 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 9
  Nov 20 17:44:36 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 14
  Nov 20 17:44:38 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 15
  Nov 20 17:44:50 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
  Nov 20 17:44:53 SERVICE-COCKPIT dhclient[281]: DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
  Nov 20 17:44:57 SERVICE-COCKPIT dhclient[281]: No DHCPOFFERS received.
  Nov 20 17:44:57 SERVICE-COCKPIT dhclient[293]: No DHCPOFFERS received.
  ```

drtyhlpr, 2018-11-21:
hm strange there seems to be no DHCP reply/offer sent from your dhcp server. can you somehow get access to the server logs?

you can use dhcping (apt-get install dhcping) to send custom dhcp requests

BrainStone, 2018-11-21:
I figured it out. And it has nothing to do with the image.  
It appears that our DHCP (Sophos Firewall XG) not only checks for the MAC-Address but also checks if the client ID or something like that is the same. And from what I understand every time the build gets a different name (like building it one day later) that ID is different. And since the Pi kept asking for an address (I kept it running) the firewall would be refusing it over and over again all while refreshing the entry in it's table.  

So I'll either disable that additional check or will just wipe the cache of the DHCP every time the issue appears again.

Thanks for pushing me in the right direction though!

closed 2018-11-21 by BrainStone