drtyhlpr/rpi23-gen-image #160: Testing needed for older Raspberry models (Pi0, Pi0W, Pi1A, Pi1B, P1B+, P1A+)

drtyhlpr, 2018-11-21:
hi testing is required for following hardware. if you own older versions of the raspberry hardware please test it with the generated stretch and buster images and provide templates if possible. it would also be nice to know about the system performance compared to the offical rpi<2 images (that should use custom "optimized"? armv6? compiler settings).

Raspberry Pi 0
Raspberry Pi 0 W
Raspberry Pi 1 model A and B
Raspberry Pi 1 model B+ and A+

thanks for your help

burnbabyburn, 2018-11-22:
armv6 with hardfloat iirc

i can do the 0W testing later on

ad65190, 2018-11-22:
Stretch booted without any issues on a Raspberry Pi 1 Model B Revision 2.0

ad65190, 2018-11-23:
Buster booted without any issues on a Raspberry Pi 1 Model B Revision 2.0

drtyhlpr, 2018-11-23:
Hi ad65190 thanks for your support. I am trying to get my hands on some older RPI0 hardware.

drtyhlpr, 2018-11-24:
I just bought a Raspberry Pi Model A. I am bussy currently will do some testing soon

drtyhlpr, 2018-12-01:
Stretch booted without any issues on a Raspberry Pi 1 Model A+ V1.1
Buster booted without any issues on a Raspberry Pi 1 Model A+ V1.1

g-vidal, 2019-01-03:
Stretch do not boot on Pi0W.
I used the template given, I get the rainbow screen but no boot. Green del flashes 2 times and nothing more, same behaviour with no screen plugged.

g-vidal, 2019-01-03:
Buster boots perfectly on Pi0W.

I must have messed up something with the stretch test. I'll retry.

Congrats @drtyhlpr  great job!

closed 2019-07-17 by drtyhlpr