drtyhlpr/rpi23-gen-image #162: cmake unable to locate

drtyhlpr, 2018-11-21:
hm strage error here:

Setting Architecture specific settings
32 bit mode selected - Setting up enviroment
Setting settings for bcm2837 Raspberry PI boards
Select DTB-File
bcm2710-rpi-3-b.dtb selected
Raspberry Pi 3 has WIFI support
the following packages needed by this script are not installed:
 cmake

do you want to install the missing packages right now? [y/n] y
E: Unable to locate package  cmake

Script done on Wed 21 Nov 2018 11:00:52 PM CET

drtyhlpr, 2018-11-21:
ps. please remove these status echos ("Setting, Select" etc) within the SET_ARCH and please remove the lines of comment signs and try to keep the comments minimal (I tried in the script so far: one comment line describing next line of code)

drtyhlpr, 2018-11-21:
maybe ENABLE_VIDEOCORE should be default false for non Xorg/GUI systems.

burnbabyburn, 2018-11-22:
> ps. please remove these status echos 

ok

> maybe ENABLE_VIDEOCORE should be default false for non Xorg/GUI systems.

good point

> E: Unable to locate package cmake

strange indeed. it's under required_packages so your host couldn't find the package. maybe have a lookt at that.
`cmake/testing,now 3.12.3-3 amd64  [installed]`

drtyhlpr, 2018-11-23:
apt-get install failed due to leading " " space inside the MISSING_PACKAGES string. hmmm

drtyhlpr, 2018-11-23:
fixed

closed 2018-11-23 by drtyhlpr