drtyhlpr/rpi23-gen-image #163: no alternatives for iptables

drtyhlpr, 2018-11-22:
```
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /root/rpi23-gen-image/images/stretch/build/chroot update-alternatives --verbose --set iptables /usr/sbin/iptables-legacy
update-alternatives: error: no alternatives for iptables
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
```

error if ENABLE_IPTABLES=true.

burnbabyburn, 2018-11-22:
i'll look at that. after i fixed the the needed bin/bash and bin/sh changes ^^

drtyhlpr, 2018-11-22:
I just learned that https://wiki.debian.org/nftables nftables is the replacement for iptables. I hope to find time soon to check and implement it.

burnbabyburn, 2018-11-22:
the following kernel features should be needed for nftables support:
        CONFIG_NETFILTER_XTABLES m
         CONFIG_NF_DUP_NETDEV m
         CONFIG_NF_NAT_SIP m
         CONFIG_NF_TABLES_ARP m
         CONFIG_NF_TABLES_BRIDGE m
         NF_TABLES m
         CONFIG_BRIDGE_EBT_BROUTE m
         CONFIG_BRIDGE_EBT_T_FILTER m
         CONFIG_BRIDGE_NF_EBTABLES m
         CONFIG_IP6_NF_IPTABLES m
         CONFIG_IP6_NF_MATCH_AH m
         CONFIG_IP6_NF_MATCH_EUI64 m
         CONFIG_IP6_NF_NAT m
         CONFIG_IP6_NF_TARGET_MASQUERADE m
         CONFIG_IP6_NF_TARGET_NPT m
         CONFIG_IP_SET_BITMAP_IPMAC m
         CONFIG_IP_SET_BITMAP_PORT m
         CONFIG_IP_SET_HASH_IP m
         CONFIG_IP_SET_HASH_IPMARK m
         CONFIG_IP_SET_HASH_IPPORT m
         CONFIG_IP_SET_HASH_IPPORTIP m
         CONFIG_IP_SET_HASH_IPPORTNET m
         CONFIG_IP_SET_HASH_MAC m
         CONFIG_IP_SET_HASH_NET m
         CONFIG_IP_SET_HASH_NETIFACE m
         CONFIG_IP_SET_HASH_NETNET m
         CONFIG_IP_SET_HASH_NETPORT m
         CONFIG_IP_SET_HASH_NETPORTNET m
         CONFIG_IP_SET_LIST_SET m
         CONFIG_NFT_BRIDGE_META m
         CONFIG_NFT_BRIDGE_REJECT m
         CONFIG_NFT_CHAIN_NAT_IPV4 m
         CONFIG_NFT_CHAIN_NAT_IPV6 m
         CONFIG_NFT_CHAIN_ROUTE_IPV4 m
         CONFIG_NFT_CHAIN_ROUTE_IPV6 m
         CONFIG_NFT_COMPAT m
         CONFIG_NFT_COUNTER m
         CONFIG_NFT_CT m
         CONFIG_NFT_DUP_IPV4 m
         CONFIG_NFT_DUP_IPV6 m
         CONFIG_NFT_DUP_NETDEV m
         CONFIG_NFT_EXTHDR m
         CONFIG_NFT_FWD_NETDEV m
         CONFIG_NFT_HASH m
         CONFIG_NFT_LIMIT m
         CONFIG_NFT_LOG m
         CONFIG_NFT_MASQ m
         CONFIG_NFT_MASQ_IPV4 m
         CONFIG_NFT_MASQ_IPV6 m
         CONFIG_NFT_META m
         CONFIG_NFT_NAT m
         CONFIG_NFT_NUMGEN m
         CONFIG_NFT_QUEUE m
         CONFIG_NFT_QUOTA m
         CONFIG_NFT_REDIR m
         CONFIG_NFT_REDIR_IPV4 m
         CONFIG_NFT_REDIR_IPV6 m
         CONFIG_NFT_REJECT m
         CONFIG_NFT_REJECT_INET m
         CONFIG_NFT_REJECT_IPV4 m
         CONFIG_NFT_REJECT_IPV6 m
         CONFIG_NFT_SET_HASH m
         CONFIG_NFT_SET_RBTREE m
         CONFIG_NF_CONNTRACK_IPV4 m
         CONFIG_NF_CONNTRACK_IPV6 m
         CONFIG_NF_DEFRAG_IPV4 m
         CONFIG_NF_DEFRAG_IPV6 m
         CONFIG_NF_DUP_IPV4 m
         CONFIG_NF_DUP_IPV6 m
         CONFIG_NF_LOG_BRIDGE m
         CONFIG_NF_LOG_IPV4 m
         CONFIG_NF_LOG_IPV6 m
         CONFIG_NF_NAT_IPV4 m
         CONFIG_NF_NAT_IPV6 m
         CONFIG_NF_NAT_MASQUERADE_IPV4 m
         CONFIG_NF_NAT_MASQUERADE_IPV6 m
         CONFIG_NF_NAT_PPTP m
         CONFIG_NF_NAT_PROTO_GRE m
         CONFIG_NF_NAT_REDIRECT m
         CONFIG_NF_NAT_SNMP_BASIC m
         CONFIG_NF_NAT_TFTP m
         CONFIG_NF_REJECT_IPV4 m
         CONFIG_NF_REJECT_IPV6 m
         CONFIG_NF_TABLES_INET m
         CONFIG_NF_TABLES_IPV4 m
         CONFIG_NF_TABLES_IPV6 m
         CONFIG_NF_TABLES_NETDEV m
         NETFILTER_XTABLES m

drtyhlpr, 2018-11-22:
omg sounds like alot of work to do hm

burnbabyburn, 2018-12-04:
see pull #170

burnbabyburn, 2018-12-07:
done in testing pls close