drtyhlpr/rpi23-gen-image #165: Kernel panic

drtyhlpr, 2018-11-22:
RPI_MODEL=3
RELEASE=stretch
BUILD_KERNEL=true
ENABLE_CRYPTFS=true
CRYPTFS_PASSWORD=...
ENABLE_VIDEOCORE=false
ENABLE_WIRELESS=true

system boots to kernel panic . needs investigation

drtyhlpr, 2018-11-22:
even without ENABLE_CRYPTFS=true, CRYPTFS_PASSWORD=... I run into a kernel panic. hm shit @burnbabyburn can you please verify?

ad65190, 2018-11-22:
I encountered the same issue with model rpi3P and both stretch and buster releases. it involved a bcm module.

burnbabyburn, 2018-11-22:
> 
> 
> even without ENABLE_CRYPTFS=true, CRYPTFS_PASSWORD=... I run into a kernel panic. hm shit @burnbabyburn can you please verify?

ofc i am

drtyhlpr, 2018-11-23:
i checked out  c5881a54a4f11dfa9b98c33e0d047515e5576a82 and do some testing now. If it works ill compare/diff the logs.

drtyhlpr, 2018-11-23:
hm fuck no. even with checkout back to https://github.com/drtyhlpr/rpi23-gen-image/commit/c5881a54a4f11dfa9b98c33e0d047515e5576a82. still kernel panic. well lets I give qemu a try.

drtyhlpr, 2018-11-23:
kernel,dtb etc. configurations are quite the same no matter if i chose RPI_MODEL 3 or 3P right?

burnbabyburn, 2018-11-23:
the kernel panic comes from a reverse partition layout on sd card

drtyhlpr, 2018-11-23:
hm? was it changed?

burnbabyburn, 2018-11-23:
not on purpose. probably some "fun" from reverting to bin/sh or escaping from shellcheck. i can boot cryptfs builds too, if i correct bootcmd to point to the right partition.

I'll revert the shellcheck changes around ROOT_OFFSET in rpi23-gen-sh and build an new one

drtyhlpr, 2018-11-23:
hm the generated images look like they have the right disk layout:

```
fdisk -l ..
2048 64M Win95 FA
231345 Linux
```

burnbabyburn, 2018-11-23:
The following line 
CMDLINE="dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2
needs to be
CMDLINE="dwc_otg.lpm_enable=0 root=/dev/mmcblk0p1
to boot my branch

drtyhlpr, 2018-11-23:
hm the 1st part should be windows to let the raspberry load and boot .txt files and bins. the root= should be the root partition of the linux partition and linux should be 2nd. hm I just removed the escapes from the shellcheck and currently building the test image

burnbabyburn, 2018-11-23:
ref+fix branch of mine successfully build

```
SET_ARCH=64
RPI_MODEL=3P
RELEASE=buster
ENABLE_CRYPTFS=true
CRYPTFS_PASSWORD=password
```

drtyhlpr, 2018-11-23:
hm i will try 64 bit now. for default 32 bit I tested to set root to 1 partition and it did not work

drtyhlpr, 2018-11-23:
`ENABLE_CONSOLE=false` should fix the current problem

burnbabyburn, 2018-11-24:
i think it's this line:
# Add serial console support
if [ "$ENABLE_CONSOLE" = true ] ; then
  CMDLINE="${CMDLINE} console=ttyAMA0,115200 kgdboc=ttyAMA0,115200"
fi
after i removed it it from cmdline, my kernel panic was gone.

burnbabyburn, 2018-12-04:
should be fixed

closed 2018-12-05 by drtyhlpr