drtyhlpr/rpi23-gen-image #171: apt-cacher-ng causing builds to fail randomly

BrainStone, 2018-12-05:
Hi. This is not a direct issue with this project. I'm simply looking for help with an issue I'm having with apt-cacher-ng, which I use to speed up the build process.

Randomly it happens that in the very early stages of the build I get errors like these:

```
[2018-12-05 08:10:41 (00:00:01.691097)] + http_proxy=http://127.0.0.1:3142/ debootstrap --arch=arm64 --foreign --components=main --include=console-data,console-setup,curl,dphys-swapfile,htop,keyboard-configuration,ncurses-term,net-tools,wget,wireless-tools,wpasupplicant,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils,locales,keyboard-configuration,console-setup,libnss-systemd,parted,dbus,openssh-server,alsa-utils,rng-tools buster /var/lib/gitlab-runner/builds/32e764b8/0/ServiceCockpit/debian-images/rpi23-gen-image/images/buster/build/chroot http://ftp.de.debian.org/debian
[2018-12-05 08:10:42 (00:00:01.783997)] I: Retrieving InRelease 
[2018-12-05 08:10:42 (00:00:02.454751)] I: Retrieving Release 
[2018-12-05 08:10:42 (00:00:02.695539)] I: Retrieving Release.gpg 
[2018-12-05 08:10:43 (00:00:03.205990)] E: Failed getting release signature file http://ftp.de.debian.org/debian/dists/buster/Release.gpg
```

```
[2018-12-05 08:02:15 (00:00:01.704736)] + http_proxy=http://127.0.0.1:3142/ debootstrap --arch=arm64 --foreign --components=main --include=console-data,console-setup,curl,dphys-swapfile,htop,keyboard-configuration,ncurses-term,net-tools,wget,wireless-tools,wpasupplicant,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils,locales,keyboard-configuration,console-setup,libnss-systemd,parted,dbus,openssh-server,alsa-utils,rng-tools buster /var/lib/gitlab-runner/builds/32e764b8/0/ServiceCockpit/debian-images/rpi23-gen-image/images/buster/build/chroot http://ftp.de.debian.org/debian
[2018-12-05 08:02:15 (00:00:01.843506)] I: Retrieving InRelease 
[2018-12-05 08:02:16 (00:00:02.785569)] I: Checking Release signature
[2018-12-05 08:02:16 (00:00:02.891257)] I: Valid Release signature (key id 16E90B3FDF65EDE3AA7F323C04EE7237B7D453EC)
[2018-12-05 08:02:18 (00:00:05.227642)] I: Retrieving Packages 
[2018-12-05 08:02:21 (00:00:07.320422)] I: Retrieving Packages 
[2018-12-05 08:02:24 (00:00:11.269990)] I: Retrieving Packages 
[2018-12-05 08:02:25 (00:00:11.574835)] E: Couldn't download dists/buster/main/binary-arm64/Packages
```

*(Note, the timestamps are added by my build system and should't affect anything really.)*

The only thing that seems to fix that temporarily is me restarting the `apt-cacher-ng` daemon. Eventually it will happen again. Usually a after a day or less.

Did anyone else have that issue? If so what can I do to fix it?

drtyhlpr, 2018-12-05:
Hi BrainStone, I did not run into issues like that using apt-cacher-ng all the time (mostly running debian in a vmware). hmmm...

do you see any interesting info inside the logs /var/log/apt-cacher-ng/apt-cacher.log or apt-cacher.err (default log location defined in /etc/apt-cacher-ng/acng.conf)

BrainStone, 2018-12-06:
I can see this in the log:

```
1544087537|E|393|127.0.0.1|debrep/dists/buster/InRelease [HTTP error, code: 503]
1544087538|I|147590|127.0.0.1|debrep/dists/buster/Release
1544087538|O|147403|127.0.0.1|debrep/dists/buster/Release
1544087538|I|2836|127.0.0.1|debrep/dists/buster/Release.gpg
1544087538|O|2656|127.0.0.1|debrep/dists/buster/Release.gpg
1544087541|E|413|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.xz [HTTP error, code: 503]
1544087544|I|10155374|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087544|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087544|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087544|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087544|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087545|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087545|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087545|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087545|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087546|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087546|O|10155168|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages.gz
1544087546|I|681|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages [HTTP error, code: 404]
1544087546|E|224|127.0.0.1|debrep/dists/buster/main/binary-arm64/Packages [HTTP error, code: 404]
1544090021|I|150004|127.0.0.1|debrep/dists/buster/InRelease
1544090021|O|149819|127.0.0.1|debrep/dists/buster/InRelease
```

And this in the err:

```
Thu Dec  6 09:12:17 2018|debrep/dists/buster/InRelease storage error [503 Inconsistent file state], last errno: File exists
Thu Dec  6 09:12:21 2018|debrep/dists/buster/main/binary-arm64/Packages.xz storage error [503 Inconsistent file state], last errno: File exists
```

A quick search brought up nothing that seemed related.
Did you change any settings or has it been just working for you out of the box?

A suggestion I'd have would be to maybe retry that stage if a cache is detected.

BrainStone, 2018-12-06:
I've never had this issue before using the cache. My connections is stable and the mirror I use is `ftp.de.debian.org`.

And to me it seems like the cache is throwing these. Not the repo itself. Especially considering after retrying 1-3 times it always works.

Maybe @burnbabyburn knows?

burnbabyburn, 2018-12-06:
from the logs i guess:

- apt-cacher-ng didn't have a packages list which was considered fresh/valid and so he asked ftp.debian.org for a new one. this is a common task to update cache and remove superseeded packages.

- from time to time even a repository is updated. ;) You probably hit an out of synch moment and got thrown out to come back later, after the sync is finished. If you use a non main mirror (e.g. a local ISP ones) you notice them more frequently, as they tend to do syncs to the main mirrors more often.

BrainStone, 2018-12-07:
I'm also getting some errors in `11apt.sh`. But I'm guessing it's related to #176

BrainStone, 2018-12-17:
Well, I'm closing this, as I can't get it to work for me. And thinking about it it's probably the fact that I'm using an Ubunut Server, but I don't know for sure. Downloading the packets doesn't take that long anyways. Much bigger savings can be achieved through `ccache`.

closed 2018-12-17 by BrainStone