drtyhlpr/rpi23-gen-image #176: APT_SERVER isn't written to /etc/apt/sources.list anymore

BrainStone, 2018-12-07:
I don't know when it happened, but currently APT_SERVER isn't written to the config anymore.  
And it seems like the lines to copy the template and adjust the repo in there have been removed.

Currently the image only contains:
```
deb http://deb.debian.org/debian buster main
```

burnbabyburn, 2018-12-07:
you really like the odd bugs :D
APT_SERVER is APT_PROXY?
Which branch? may i assume dev branch?

BrainStone, 2018-12-08:
I couldn't find the commit, but in the older versions it was written to the file. I'll try to find the commit on mondy when I have access to my work files.

drtyhlpr, 2018-12-09:
hi, found it here:

https://github.com/drtyhlpr/rpi23-gen-image/commit/a90f6ad1a83f24641bd13c5283826f87aea6d945#diff-1b4928ad72fffd456ece9e3ff741bb8a

`else # BUILD_KERNEL=true` block was accidentally deleted it seems :)

drtyhlpr, 2018-12-09:
quick fix:
https://github.com/drtyhlpr/rpi23-gen-image/commit/be99731b278c6a6944df32fd037bbbd068b74203

burnbabyburn, 2018-12-09:
thx guys :)

drtyhlpr, 2018-12-09:
should be fixed

closed 2018-12-09 by drtyhlpr