drtyhlpr/rpi23-gen-image #178: Overlays

burnbabyburn, 2018-12-12:
Am i the only one who is missing the overlays under `/boot/overlay`?
I needed to add an `o` to `*.dtb` in `13-kernel.sh ` - after that the overlays are beeing copied
the qemu part in `rpi23-gen-image.sh` would need an `o` too.

i checked with raspian and they also contain `dtbo `instead of `dtb `overlays.
```
 # Copy compiled dtb device tree files
  if [ -d "${KERNEL_DIR}/arch/${KERNEL_ARCH}/boot/dts/overlays" ] ; then
    for dtb in "${KERNEL_DIR}/arch/${KERNEL_ARCH}/boot/dts/overlays/"*.dtbo ; do
      if [ -f "${dtb}" ] ; then
        install_readonly "${dtb}" "${BOOT_DIR}/overlays/"
      fi
    done
```

if you can confirm this issue ill make a pull

drtyhlpr, 2018-12-14:
hi, I can not verify currently. If I remember right I recently added this if statements. hmmm "make firmware" or something like that was removed from the latest kernel versions if i remember right and the "if overlays" checck was implemented because of that hm but I don't remember more...

burnbabyburn, 2018-12-16:
files got to /boot/firmware/overlay
maybe they changed the overlay file ending, maybe they didn't :) dtbo is the correct one ;)

closed 2018-12-16 by burnbabyburn