drtyhlpr/rpi23-gen-image #179: rc.local: touch: cannot touch '/run/systemd/resolve/resolv.conf': No such file or directory

BrainStone, 2018-12-14:
The rc.firstboot script errors with the following issue:

```
Jun 22 11:12:00 SERVICE-COCKPIT rc.local[254]: touch: cannot touch '/run/systemd/resolve/resolv.conf': No such file or directory
Jun 22 11:12:00 SERVICE-COCKPIT systemd[1]: rc-local.service: Control process exited, code=exited status=1
Jun 22 11:12:00 SERVICE-COCKPIT systemd[1]: rc-local.service: Failed with result 'exit-code'.
Jun 22 11:12:00 SERVICE-COCKPIT systemd[1]: Failed to start /etc/rc.local Compatibility.
```

I am using the testing branch.

BrainStone, 2018-12-17:
Second update, the file is not in the image, which makes it even weirder. Something must create during the first boot before the `/etc/rc.firstboot` script executes.  
That means most likely something I'm doing must add it if none of you have the same issue.

Will keep you posted!

burnbabyburn, 2018-12-18:
I'll watch for this error on my next install. I didn't notice smth. special as `systemd-resolved` and `/etc/resolv.conf` are in my image atm.

However i did experience this error once. It was a prepending error in rc.firstboot which caused it.

Btw: While looking at the referenced line - This looks wierd to me. Both `-a` and `-e` do check if file exists and one is negated. But script doesn't break so it seems atleast valid :D

BrainStone, 2018-12-18:
The `-a` is an `AND` in that situation.

And no it doesn't. As I said, the first check fails.
And the file should've been removed from the image: https://github.com/drtyhlpr/rpi23-gen-image/blob/testing/rpi23-gen-image.sh#L693

BrainStone, 2018-12-18:
Also I just tested removing the file `/etc/resolv.conf`, then rebooting. The script still failed at the same spot, so something is actively adding the file before `rc.firstboot` is run.

BrainStone, 2018-12-18:
I suggest  either just removing the check for `/etc/resolv.cong` or additionally also removing the file entirely, so it becomes that symlink in any case.

BrainStone, 2018-12-18:
@burnbabyburn do you also happen to have `dhcpcd` added to your image?

BrainStone, 2018-12-19:
Ok. I disabled the service before first boot and the issue went away, so it's `dhcpcd` for sure.

I'll soon create a PR that will improve the wireless stuff. That will also include the fix, as I only even have that service for WiFi.

closed 2019-07-17 by drtyhlpr