drtyhlpr/rpi23-gen-image #183: Dependency pb on rpi3B+ buster-arm64 build on libpam-systemd

g-vidal, 2018-12-27:
Hi all,

Building arm64 stretch rpi3B+ works perfectly thanks for the job, but I have tried to build a buster image for rpi3B+ and it failed on systemd-shim dependency of libpam-systemd which is not(yet) available in buster. This package is available in stretch and sid.
Would it be possible to have a configuration 

> buster+sid

  that would enable to build successfully a buster image by picking the missing packages in sid. I suppose that it might be possible to configure the build in such a way but it is unfortunately beyond my skills.

g-vidal, 2018-12-29:
Found a workaround...
The dependency is on systemd-shim or systemd-sysv  but by default the configuration provides :
`ENABLE_SYSVINIT=false`
which prevents the installation of systemd-sysv by changing to 
`ENABLE_SYSVINIT=true`
The build is successfull. 
I have not yet checked if the raspi is fully operationnal, I'll provide info if not.

burnbabyburn, 2019-01-24:
thx for reporting this issue @g-vidal 

did you encounter any errors while operating your pi?
i don't know the systemd-shim package. but if you add the debian sid repo to apt sources it **may** be possible to install packages with via APT_INCLUDE=systemd-shim/unstable

AFAIR APT_INCLUDE just forwards to apt install

burnbabyburn, 2020-02-17:
@drtyhlpr
close?

g-vidal, 2020-02-20:
Everything runs OK 
you can close ith maybe adding an alert on the README  for that option and anyway  the migration is progressing and fewer people are attached to systemd-sysv workarounds
thanks

closed 2020-02-20 by g-vidal