drtyhlpr/rpi23-gen-image #194: build kernel - /bin/sh: 1: bison: not found

tswaehn, 2019-03-17:
I created a fresh debian x64 VM on virtualbox, cloned the rpi23-gen-image and tried to create an image for RPI3 stretch (all settings default)

(first minor issue was, that Qemu was not installed automatically - fixed that manually)

but second issue is the one below:

```
make -C /home/siri/tmp/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make: Entering directory '/home/siri/tmp/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
/bin/sh: 1: bison: not found
scripts/Makefile.lib:196: recipe for target 'scripts/kconfig/zconf.tab.c' failed
make[1]: *** [scripts/kconfig/zconf.tab.c] Error 127
Makefile:539: recipe for target 'bcm2709_defconfig' failed
make: *** [bcm2709_defconfig] Error 2
```

tswaehn, 2019-03-17:
looks like manual install of bison solves the job
```
sudo apt-get install bison
```
Actually flex was missing too
```
sudo apt-get install flex
```

tswaehn, 2019-03-17:
Question: is it expected to install these manually or shall we check and install automatically?

tswaehn, 2019-03-17:
if needed, we could modify this line
```
REQUIRED_PACKAGES="debootstrap debian-archive-keyring qemu-user-static binfmt-support dosfstools rsync bmap-tools whois git bc psmisc dbus sudo"
```
and add ```flex,bison```

quozl, 2019-03-17:
Kernel package build dependencies change over time as the kernel developers change the configuration.  Make sure you have `deb-src` entries in `sources.list` and use `apt build-dep KERNEL` where KERNEL is the name of a recent kernel package.

tswaehn, 2019-10-08:
I just did setup a fresh debian and installed all "kernel depencies" as per you suggetion 

`apt-get build-dep linux-image-XXXX` 

and it did the trick.

However, for me speaking that step is missing in the README.md

Is there a general way of adding this to the README.md ?

burnbabyburn, 2019-10-09:
> Question: is it expected to install these manually or shall we check and install automatically?

it is unexpected. If `flex `and `bison `are now required for default kernel, add it to the `REQUIRED_PACKAGES`.
Last time i checked only `nexmon `and `uboot `option required those packages.

Could you create a pull request?

Thanks/Danke!

tswaehn, 2019-12-09:
Created pull request

drtyhlpr, 2019-12-09:
fixed!

closed 2019-12-09 by drtyhlpr