drtyhlpr/rpi23-gen-image #195: build kernel - fatal error: openssl/bio.h: No such file or directory

tswaehn, 2019-03-17:
AFAIK: openSSL is not part of the kernel. still it tries to build a user module.

```
  HOSTCC  scripts/extract-cert
scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory
 #include <openssl/bio.h>
                         ^
compilation terminated.
scripts/Makefile.host:90: recipe for target 'scripts/extract-cert' failed
make[1]: *** [scripts/extract-cert] Error 1
make[1]: *** Waiting for unfinished jobs....
Makefile:1062: recipe for target 'scripts' failed
make: *** [scripts] Error 2
make: Leaving directory '/home/siri/tmp/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux'
```

tswaehn, 2019-03-17:
it feels like ```libssl-dev``` on the host is missing

tswaehn, 2019-03-18:
it feels like ```libssl-dev``` on the host is missing and does the trick

tswaehn, 2019-12-09:
created pull request #210

drtyhlpr, 2019-12-09:
fixed!

closed 2019-12-09 by drtyhlpr