drtyhlpr/rpi23-gen-image #197: Seg fault

Xanibod, 2019-03-20:
Having a problem during the building of the kernel.
Both of these commands,
```
sudo CONFIG_TEMPLATE=rpi3-stretch-arm64-4.14.y ./rpi23-gen-image.sh
sudo SET_ARCH=64 RPI_MODEL=3 RELEASE=stretch BUILD_KERNEL=true ./rpi23-gen-image.sh
```

results in this:

```
  CC      scripts/mod/devicetable-offsets.s
scripts/mod/devicetable-offsets.c: In function ‘main’:
scripts/mod/devicetable-offsets.c:229:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
scripts/Makefile.build:131: recipe for target 'scripts/mod/devicetable-offsets.s' failed
make[2]: *** [scripts/mod/devicetable-offsets.s] Error 1
scripts/Makefile.build:544: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
  LEX     scripts/genksyms/lex.lex.c
  YACC    scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  YACC    scripts/dtc/dtc-parser.tab.h
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.c
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
Makefile:1062: recipe for target 'scripts' failed
make: *** [scripts] Error 2
```

However,
`sudo CONFIG_TEMPLATE=rpi3stretch ./rpi23-gen-image.sh`
completes without issues.

burnbabyburn, 2019-06-27:
if the error persists and it was not just a faulty commit, pls try 4.19

drtyhlpr, 2019-07-17:
Please reopen issue if the problem still exists

closed 2019-07-17 by drtyhlpr