drtyhlpr/rpi23-gen-image #202: Does the RPI4 run with the 32bit buster image?

drtyhlpr, 2019-07-17:
hi, can anyone with a rpi4 test if the 32bit buster image is running on it?

thanks for your help

drtyhlpr, 2019-07-19:
I added rpi4 model support patches, hf
https://github.com/drtyhlpr/rpi23-gen-image/commit/7319d292c44d7ec1dd3bf7b1f9caf07894bccaf6

ljones0, 2019-07-21:
Not completely tried everything but I left SET_ARCH at 32 bits and set RPI_MODEL to 4 but it did not boot at all on a PI4 (also tried different micro sd cards). I tried to SET_ARCH to 64 bits and its gotten stuck at "I: Configuring man-db".

ljones

drtyhlpr, 2019-07-21:
hi thanks for you feedback, I ll take a look at the boot and compare with rasberian offical images etc.

g-vidal, 2019-11-23:
Hi all,
Succeeded to compile buster/32 but  nothing when plugin the rpi after installing the card. red light switches on green flashes 3 times then nothing more ;-(  I have connected to a screen and no signal I tried HDMI 10 and HDM 11 connector... 
Ready to test if any progress on your side.
Thanks for the work.

g-vidal, 2020-02-04:
Answer is YES after the two pullrequests have been made B-)