drtyhlpr/rpi23-gen-image #203: Permission problems with rpi23genimage

ljones0, 2019-07-26:
Hello,

Getting a new problem generating any image with rpi23genimage now unfortunatly. Wanted to give this script a go on my pi zero and pi 3b+ but tried on both and also both distros (configured in the build script) but I am getting the same error every time now;

[code]
+ as_nobody mktemp -d
+ sudo -E -u nobody LANG=C LC_ALL=C mktemp -d
sudo: unable to resolve host debian
+ temp_dir=/tmp/tmp.59pXN4MDXB
+ as_nobody git -C /tmp/tmp.59pXN4MDXB clone https://github.com/RPi-Distro/pi-bluetooth.git
+ sudo -E -u nobody LANG=C LC_ALL=C git -C /tmp/tmp.59pXN4MDXB clone https://github.com/RPi-Distro/pi-bluetooth.git
sudo: unable to resolve host debian
Cloning into 'pi-bluetooth'...
remote: Enumerating objects: 100, done.
remote: Total 100 (delta 0), reused 0 (delta 0), pack-reused 100
Receiving objects: 100% (100/100), 12.62 KiB | 0 bytes/s, done.
Resolving deltas: 100% (33/33), done.
warning: unable to access '/root/.config/git/attributes': Permission denied
+ mv /tmp/tmp.59pXN4MDXB/pi-bluetooth /media/aaaaa/ramdisk/rpi23-gen-image-master/images/stretch/build/chroot/tmp/
+ as_nobody wget -q -O /media/aaaaa/ramdisk/rpi23-gen-image-master/images/stretch/build/chroot/tmp/pi-bluetooth/LICENCE.broadcom_bcm43xx https://aur.archlinux.org/cgit/aur.git/plain/LICENCE.broadcom_bcm43xx?h=pi-bluetooth
+ sudo -E -u nobody LANG=C LC_ALL=C wget -q -O /media/aaaaa/ramdisk/rpi23-gen-image-master/images/stretch/build/chroot/tmp/pi-bluetooth/LICENCE.broadcom_bcm43xx https://aur.archlinux.org/cgit/aur.git/plain/LICENCE.broadcom_bcm43xx?h=pi-bluetooth
sudo: unable to resolve host debian
/media/aaaaa/ramdisk/rpi23-gen-image-master/images/stretch/build/chroot/tmp/pi-bluetooth/LICENCE.broadcom_bcm43xx: Permission denied
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
[/code]

Not sure what is causing it - maybe a problem somehow compiling the bluetooth software?

ljones

tyborall, 2019-08-25:
I have exact the same issue and are not able to build the image therefore. I build on a Debian 10 Buster as host system. These are the last lines of build log before it ends:

[image]

burnbabyburn, 2019-10-09:
1. imho code looks fine. should be fixed since january https://github.com/drtyhlpr/rpi23-gen-image/commit/c0630e37f6c3fa8ca1502372f9c09992c9382581
2. Licence file is still up

I guess this could be a config problem?

`warning: unable to access '/root/.config/git/attributes': Permission denied`
https://stackoverflow.com/questions/27150926/unable-to-access-git-attributes

or this
```
sudo -E -u nobody LANG=C LC_ALL=C wget -q -O /media/aaaaa/ramdisk/rpi23-gen-image-master/images/stretch/build/chroot/tmp/pi-bluetooth/LICENCE.broadcom_bcm43xx https://aur.archlinux.org/cgit/aur.git/plain/LICENCE.broadcom_bcm43xx?h=pi-bluetooth
sudo: unable to resolve host debian
```
https://askubuntu.com/questions/59458/error-message-sudo-unable-to-resolve-host-none

burnbabyburn, 2020-02-17:
@drtyhlpr 
since its fixed i suggest close?