drtyhlpr/rpi23-gen-image #207: feat: custom.d firstboot action

tswaehn, 2019-11-17:
I was wondering if we could implement a standardized hook for firstboot actions. 

why?:

- actions (ex. enable services) that need to be performed after all the default firstboot is done should be added

suggestions:

- we add by calling predefined script in ./custom.d/firstboot/

this does basically the same as the default firstboot:

> cat files/firstboot/10_after_boot.sh >> "${ETC_DIR}/rc.firstboot"

drtyhlpr, 2019-11-20:
Hi, feel free to add the stuff that you think is usefull for you and others :)  sounds good!

tswaehn, 2019-11-20:
@drtyhlpr thanks. I try to not interfere with any design paths :)

burnbabyburn, 2020-02-15:
Kannst du mir das genauer erklären? Ich wollte das gerade in Code umsetzen und verstehe nicht was du meinst.

VonOx, 2020-04-09:
Hi

Need this feature too
Just handle firstboot script in custom.d if exist

@burnbabyburn  or @tswaehn , have you made the modification ? If not i will make a PR

commit 55670e66 (2020-04-12): Merge pull request #220 from VonOx/custom-firstboot

Fix #207 Update 50-firstboot.sh to handle custom firstboot scripts - thanks to VonOx for the fix!
closed 2020-04-12 by drtyhlpr