drtyhlpr/rpi23-gen-image #208: error msg: bcm2708-rng -- Failed to find module 'bcm2708_rng'

tswaehn, 2019-11-17:
I do have

- stretch 32bit @ RPI 3b+

I am getting this when booting.

As far as I understand the **bcm2708-rng** is depricated and replaced by **bcm2835_rng**. However it gets somehow pulled in. Right now I dont have any clue yet, to get rid of this.

[https://archlinuxarm.org/forum/viewtopic.php?f=64&t=10153](reference)

g-vidal, 2019-12-13:
HI I am wondering if it is  the same problem than  #209 and #202 the bootstrapping does not retrieve the up-to-date binaries from [raspberrypi/linux](https://github.com/raspberrypi/linux).
It seems that on the site the files with new numbers are avalable  and are not catched.

matyapiro31, 2020-02-11:
You can remove it by editing `/lib/modules-load.d/rpi2.conf`.

burnbabyburn, 2020-04-15:
We should be fine with just the dtb overlays and options nowaydays. Have to test, espacially with older releases
https://www.raspberrypi.org/forums/viewtopic.php?t=98318
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README