drtyhlpr/rpi23-gen-image #209: Crash compiling Pi3-Bullseye 32 missing script mkknlimg kernel 5.3 and up

g-vidal, 2019-12-02:
Hi all,
I am compiling  a Pi3 under bullseye using kernels from [paspberrypilinux](https://github.com/raspberrypi/linux) and I get the folowing message :
```
+ /data/RpiGenImage/Images/bullseye/build/chroot/usr/src/linux/scripts/mkknlimg /data/RpiGenImage/Images/bullseye/build/chroot/usr/src/linux/arch/arm/boot/zImage /data/RpiGenImage/Images/bullseye/build/chroot/boot/firmware/kernel7.img
./rpi23-gen-image.sh: 517: bootstrap.d/13-kernel.sh: /data/RpiGenImage/Images/bullseye/build/chroot/usr/src/linux/scripts/mkknlimg: not found

```
I have checked the file is not present on versions 5.3 and 5.4 (5.2 has got it). Is it missing by accident or is there a change in the process after kernel version 5.2 ?

I would have liked  to download the file and check what happens if I put it in the  proper folder but I don't know how to resume at the point the preparation crashed. Could it be possible ? How?
thanks

drtyhlpr, 2019-12-08:
hi g-vidal.

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/scripts/mkknlimg
part of the raspberry kernel scripts dir  (here is a note that the tool moved to the kernel scripts dir https://github.com/raspberrypi/tools/tree/master/mkimage)

i ll check

g-vidal, 2020-02-04:
I replaced  `mkknlimg`  simply by `cp` as proposed in one of the comments on te raspberry/linux thread. Seems to work like a charm.
Can you pull the request?