drtyhlpr/rpi23-gen-image #214: Rpi4 64 bit image not working

ljones0, 2020-02-10:
Hello!

Had a problem trying to create a 64 bit image for the raspberry pi 4 - the image did not boot at all. But partway through I saw the following;

[code]
+ make -C /media/ramdisk/rpi23-gen-image-master/images/buster/build/chroot/usr/src/linux -j12 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- CC=aarch64-linux-gnu-gcc Image dtbs
make: Entering directory '/media/ramdisk/rpi23-gen-image-master/images/buster/build/chroot/usr/src/linux'
scripts/kconfig/conf  --syncconfig Kconfig
.config:1188:warning: symbol value 'm' invalid for NF_TABLES_BRIDGE
.config:2467:warning: symbol value 'm' invalid for MT76_LEDS
.config:3849:warning: symbol value 'm' invalid for SMS_SIANO_RC
.config:6992:warning: override: reassigning to symbol KEY_DH_OPERATIONS
.config:6993:warning: override: reassigning to symbol BCM_VC_SM
.config:6993:warning: symbol value 'm' invalid for BCM_VC_SM
.config:6994:warning: override: reassigning to symbol TCG_TIS
.config:6995:warning: override: reassigning to symbol DRM_VC4_HDMI_CEC
.config:6997:warning: override: reassigning to symbol CEPH_LIB
.config:6999:warning: override: reassigning to symbol CEPH_FS
.config:7002:warning: override: reassigning to symbol BLK_DEV_RBD
.config:7003:warning: override: reassigning to symbol CPU_FREQ_DEFAULT_GOV_ONDEMAND
*
* Restart config...
*
*
* Ethernet Bridge nf_tables support
[/code]

Not sure about those warnings and then it suddenly ended up going into an interactive kernel configuration.

ljones

g-vidal, 2020-02-11:
Hi,
I compiled a buster 64 yesterday with a kernel 5.4 without problem (with the 2 pull requests I have propose included) and it boots perfectly. Maybe an issue with some options or choices.  I accept the nf_tables option  and it is not compiled as a module maybe it is the cause of the warnings you are quoting....
I had the same issue and in my case when the pi4 was not booting it was due to the missing instructions decribed in #212 , after addind the lines proposed everything was OK. Please try it that may confirm my pul request.

Thanks

ljones0, 2020-02-11:
ok I gave the files at https://github.com/g-vidal/rpi23-gen-image (#212) but I still got the same messages regarding nf_tables and the pi still didn't want to boot x.x

burnbabyburn, 2020-02-17:
the override in kernel config is ok most of the time. do i understand correctly: the image compiles fine but does not boot?

> ok I gave the files at https://github.com/g-vidal/rpi23-gen-image (#212) but I still got the same messages regarding nf_tables and the pi still didn't want to boot x.x

the override in kernel config is ok most of the time. do i understand correctly: the image compiles fine but does not boot? would you post your config?

ljones0, 2020-03-06:
>the override in kernel config is ok most of the time. do i understand correctly: the image compiles fine but does not boot? would you post your config?

Wouldn't the config file be too long to post here? BTW you are correct; the image compiles fine but does not boot.

>did you enable KERNEL_MENUCONFIG and set it to true instead of false?
Nope, didn't change that at all I did not touch it.

ljones

burnbabyburn, 2020-03-12:
>   .config:1188⚠️ symbol value 'm' invalid for NF_TABLES_BRIDGE
>   .config:2467⚠️ symbol value 'm' invalid for MT76_LEDS
>   .config:3849⚠️ symbol value 'm' invalid for SMS_SIANO_RC
>   .config:6993⚠️ symbol value 'm' invalid for BCM_VC_SM
you could remove or replace with y/n and compile again to test if its a problem with those settings.
change e.g.	    `set_kernel_config CONFIG_BCM_VC_SM m` to 	    `set_kernel_config CONFIG_BCM_VC_SM y` or remove parameter completly in [ "$RPI_MODEL" = 4 ] section in 13-kernel.sh

I don't have a pi4 to test this.

when your pi is not booting, do you see rainbow screen or does it stay black?

ljones0, 2020-04-12:
Ok I tried again but same problem, no boot at all. I did notice that the pi's green LED dosen't even light.

I would attach a log, but it is too large to attach here - the log file is 1.1mb in size!

ljones

ljones0, 2020-04-12:
[build.log](https://github.com/drtyhlpr/rpi23-gen-image/files/4466194/build.log)

Tried to upload a log file there but being the size that it is I'm not sure if it is complete or will even work.

ljones