drtyhlpr/rpi23-gen-image #217: All generated images freeze on the rainbow screen

XIIIVI, 2020-03-05:
Hi all,

First of all, congratulations for your awesome work.
Thanks to it, we managed to divide by 2 the size of our images !

Anyway, I had many issues when using the script from the master of this project whereas the version coming from https://github.com/g-vidal/rpi23-gen-image works like a charm.
The latest one integrates the pull request consisting of copying more files in the script 15-rpi-config.sh and ... maybe something more as I tested the patched script in this github project without success.

All the generated images freeze on the rainbow script when booting.

Best regards,

VonOx, 2020-04-13:
I've the same problem

burnbabyburn, 2020-04-16:
ordered a pi4 today. ill be back!

VonOx, 2021-02-22:
This issue is solved , right ?

g-vidal, 2021-02-22:
I think so. I produced a series of operational images around the new year without problem.