drtyhlpr/rpi23-gen-image #228: Pi4 and Pi4-64 Buster ans Bullseye are back ! (after #226 and #227)

g-vidal, 2020-11-19:
I have successfully created Pi4-Buster Pi4-64 Buster and Pi4-64 Bullseye.
Images are available from this [web page](http://mediaserv.tremplin.ens-lyon.fr/isoImages/RaspBerry/), comments are  in french, (sorry about that I have had not enough time for translations until now).

burnbabyburn, 2021-01-03:
thx for providing some precompiled images vidal!

zylo117, 2021-03-01:
thanks for the hard work! Can you provide the pi4 64 bit links again? The link in your website is broken.
http://mediaserv.tremplin.ens-lyon.fr/isoImages/RaspBerry/DebianBusterPi4/22021-02-04-rpi4-64-buster.img.xz

g-vidal, 2021-03-01:
I am very sorry , I have changed the structure of my repository and missed one of the movements. Things should be OK now.  Let me know if any problem.
Enjoy!

g-vidal, 2021-11-07:
Good news for you @zylo117  and maybe others.... I have compiled with rpi23-gen-image on kernel 5.15.y an image for : 
* raspberry Pi3
* raspberry Pi3B+
* raspberry Pi4
* raspberry Pi4_64
You can find them  on [Tremplin server](http://mediaserv.tremplin.ens-lyon.fr/isoImages/RaspBerry/). All the building process and parameters used can be found on [my blog](https://blog.tremplin.ens-lyon.fr/GerardVidal/RaspberryPiBuildImage.html) (still in french sorry about that but pieces of code are international ;-) ).
There might be some mistakes here and there please  let me know if you detect one. As these images are built for a special purpose you will find inside some goodies : 
```
* lxdm lxqt lxde
* vncserver on :5901 carried by user  tsuser
* xrdp
* jupyterhub carried by user callisto from a virtualenv (with jupyter python3 java and javascript engines)
* mraa and upm (eclipse tools to handle GPIO and many sensors)
* weewx carried by user weewx (management tool for weatherstations)
* domotix
* kodi
* Snap (self hosted free version of scratch-like software)
```